sy2002/QNICE-FPGA #131: GitHub Repo Languages

sy2002, 2020-09-14:
The current display of programming languages used in our GitHub repo is just plain wrong:

[image]

I assume this is a bug in GitHub's Linguist:

https://github.com/github/linguist

**This problem is worst than meets the eye: GitHub's AI is showing interessting repos to other users based on the user's experiene and own repos. Our repo will never be shown to any VHDL savvy person but only to HTML geeks** :rage1:

We should solve that for V1.7

Linguist allows specifying what shall be shown in some file that you need to check-in.

MJoergen, 2020-09-15:
I've added a .gitattributes file to the repository and fixed the worst errors.

sy2002, 2020-09-15:
Great job, Michael: I checked in this file to our `master` branch. (Checking something in to our `master` branch outside a release is something, that I am frowning upon very much. But this was an emergency.) Sucess: Look at this new graph:

[image]

So I closed this issue.

closed 2020-09-15 by sy2002