sy2002/QNICE-FPGA #132: Disassembled RBRA looks odd on VGA

sy2002, 2020-09-14:
On VGA, the Monitor's new feature to disassemble `RBRA` in a convenient way look odd on VGA. I guess is due to the usage of tabs. See image:

[image]

There are multiple ways to solve that:

* We could add support for `\t` in the Monitor's VGA output routines. (We would need to decide how many spaces)
* We could change the Monitor's disassembler to use a dynamic amount of spaces so that the output still is nicely formatted one column below each other without thrashing
* <etc.>

Up to the group to discuss how to proceed here.

bernd-ulmann, 2020-09-14:
When I extended the assembler for \t in ASCII strings I went a not too-clever route to implement this feature which most probably is the reason for this ugly output (lines 686 - 689 in qasm.c). A \t control sequence basically gets replaced by a SPACE (ugly, it was a pretty quick fix) and an ASCII 9, i.e. a "real" tab control character. 

I assume that this is displayed as a thick vertical bar in the output. The VGA output routines should take care of tab control characters.

Regarding spacing: The standard is a tab stop every eight columns. We should keep that - maybe in a configurable way, what do you think?

sy2002, 2020-09-14:
OK got it - then we will enhance the VGA routines in the monitor.

Just out of curiosity: why did you add the space and not just use the "pure" tab character?

@MJoergen should I assign this one to you or me?

bernd-ulmann, 2020-09-14:
The reason for 0x0a 0x09 is simple: Because I did not find a clean way to do so. ;-) If you can fix this easily, please do so. :-)

MJoergen, 2020-09-14:
@sy2002 . I think it is faster if you assign it to yourself :-)

sy2002, 2020-09-15:
OK. Done :-)

bernd-ulmann, 2020-09-15:
I just changed the assembler (its gets more and more ugly... *ARGL*) so that \t is now translated into 0x09 instead of 0x09 0x20 as before. Thus we got rid of the erroneous blank character.

sy2002, 2020-09-15:
Hooraaay :-)