sy2002/QNICE-FPGA #133: Finalize doc/constraints.md

sy2002, 2020-09-15:
@MJoergen Please perform these tasks and assign the issue to me afterwards:

* Please add the exact USB keyboard and SD card type/model you are using in constraints.md in the appropriate section. We are "collecting" known to work SD cards and keyboards because by far not all do.

* Please read through the VBCC constraints and check, if you discovered more constraints (not to be mixed up with programming best practices, but more like real constraints)

* After that please assign this issue to me and I will perform the following tasks

Tasks that I will perform:

* Sift through TODO.txt and other notes I have scattered around and collect *really all* known constraints and document them in `constraints.md`

* Finalize all "TODO" sections that are currently still there in `constraints.md`

MJoergen, 2020-09-15:
Done.

sy2002, 2020-09-15:
@MJoergen Thank you. Can you also add the exact SD card model (including brand, class, size, model) to the list?

MJoergen, 2020-10-03:
Maybe we should re-test with different SD cards. I just checked, and it works with my class 10 card.

sy2002, 2020-10-03:
@MJoergen Would be great! But be aware: Our today's implementation sucks. The reason is, that the SPI mode that we are using is not really well implemented by the various SD card makers. If we (reads like "you" :-)) would refactor the SD component to use the *real* *native* SD card mode, then we would support all cards like a charm.

I googled and found a spec here. No idea, if it is complete, but it is freely available on the web:

https://www.convict.lu/pdf/ProdManualSDCardv1.9.pdf