sy2002/QNICE-FPGA #135: Introduce Base Pointer Shadow Register for Register Banks

sy2002, 2020-09-15:
In a phone call, Bernd and I discussed the following new feature:

* Introduce "Base Pointer" shadow register for register banks called `RB` (aka "Register Basepointer")
* The shadow register will be accessed via our newly defined `RSR` (read shadow register) and WSR (`write shadow register`) instructions (see also issue #127)
* The value of the Base Pointer is always automatically added to the upper 8 bits of the SR so that the currently selected register bank is always `RB + SR(15 downto 8)`

Use cases for this:

* ISRs could fairly assume, that from register bank 128 on, the banks are theirs, set BP to 128 and after that do their INCRB/DECRB within the ISR. This would solve the challenges described in https://github.com/sy2002/QNICE-FPGA/issues/109#issuecomment-691921208, https://github.com/sy2002/QNICE-FPGA/issues/109#issuecomment-691933091 and https://github.com/sy2002/QNICE-FPGA/issues/109#issuecomment-691950116

* Normal functions could reserve some space on the "register bank stack" by bumping up the base pointer and then safely access this "reserved space"

MJoergen, 2020-09-17:
Just a thought: Currently, the CPU preserves the values of `R13`, `R14`, and `R15` upon entry into an interrupt routine. These values are then restored when an `RTI` is executed.

This means the ISR is free to manipulate the register bank AND DOES NOT NEED TO RESTORE THIS. I just verified this by removing the last `DECRB` from the ISR in the `fancy.asm`. I.e. I replaced

```
DECRB
RTI
```

with

```
RTI
```

This is a "mismatch", in the sense that the ISR no longer preserves the register bank. But this does not matter, because the CPU does this automatically. I verified on hardware that everything works as before without this explicit `DECRB` instruction.

I believe this observation invalidates at least some of the arguments for a Register Bank Base Point register. Instead, an ISR could just manually set the register bank to, say, 128 upon entry, i.e. by beginning with an instruction like `MOVE 0x8000, R14`. The ISR can then do subroutine calls and use both the register banks and the stack as we do now. And regular programs can safely rely on, that all register banks below 128 are never destroyed.

So I just verified the above by replacing the first instruction of the `fancy` ISR (`INCRB`) with the instruction `MOVE 0x8000, R14`. Everything still works as before! Furthermore, the CPU test runs without problems even while the fancy ISR is running! Only change needed in the CPU test was that it tests all register banks, and I changed it to only test the first 128 register banks.

I must admit I don't really like the idea of a Register Bank Base Pointer, and I believe we should think some more about whether we actually need them.

bernd-ulmann, 2020-09-17:
Good point! :-) 

I am not too fond of a register bank base pointer as it, too, clutters the architecture which becomes more and more clumsy.

Let's talk about this on Saturday as it fits well into the topic of how to access the shadow registers (if necessary).

sy2002, 2020-09-17:
I find Michaels https://github.com/sy2002/QNICE-FPGA/issues/135#issuecomment-693927113 absolutely brilliant.

It elegantly solves all issues that I had and lets me continue to reserve memory in the regbank (as long as I exactly know what I am doing) 😈 

So I am joining Michaels and Bernds opinion that we an focus on Saturday to the shadow register use case of hardware single step debugging and we can drop the idea of a regbank base pointer. I am closing this issue.

closed 2020-09-17 by sy2002