sy2002/QNICE-FPGA #136: New semantics of our interrupt devices

sy2002, 2020-09-15:
* We need to document these semantics somewhere (possible interrupt device documentation, best practice programming, ...)
* The timer will be disabled if any of the two registers prescaler or timer is zero.
* The VGA scanline interrupt will be disabled, if bit 15 of the scanline register is set.
* Test programs (timer test, fancy, ...) need to be adjusted
* Emulator needs to be adjusted

sy2002, 2020-09-17:
We must not forget to adjust the assembler and C test programs to adhere to these new semantics and to our new ISR best practices. I adjusted `test_programs/fancy.asm` but all the others are still adhering to the old semantics.

sy2002, 2020-09-19:
We also must not forget to adjust all assembler and C test programs to activate interrupts at first because interrupts are now disabled by default (see https://github.com/sy2002/QNICE-FPGA/issues/139#issuecomment-694474689)

MJoergen, 2020-09-19:
I suggest the Monitor enables global interrupt.

sy2002, 2020-09-19:
@MJoergen Can you take this one and finalize it? The remaining TODOs as far as I oversee them correctly are:

- [x] Hardware: Adjust the **VGA** to the new semantics
- [ ] Hardware: Adjust the **timer module** to the new semantics and also adjust the comments in the two file's headers as they describe the old semantics
- [ ] Make sure the Monitor enables the global interrupt
- [x] Adjust `fancy` to the new semantics
- [ ] Adjust all other interrupt test and demo programs in assembler and in C