sy2002/QNICE-FPGA #139: Ability to stop/enable interrupt processing via a very simple interrupt controller

sy2002, 2020-09-17:
While discussion issue #138 the need for being able to stop/enable interrupt processing came up.

To prevent a further complication of QNICE's ISA, the idea is:

* Let's make a very (!) simple interrupt controller device: One register, where you can use one bit to enable/disable all interrupts.
* The interrupt controller is put between the CPU and the first Daisy-chained device
* (tbd) It combinatorially connects the enable bit with the INT_N line and it just passes through the GRANT_N line

In our discussion we came to the conclusion, that this mechanism works free of race-conditions as long as we are not having a pipelined CPU.

MJoergen, 2020-09-17:
So this is the same functionality as the 6502 instructions CLI and SEI, right ?

Why not place this functionality inside the CPU ? I.e. have the CPU keep a bit indicating whether interrupts are globally enabled or disabled. This could be one of the bits in R14.

Another - somewhat different - solution is to make use of the INT instruction (i.e. software interrupts) for our SYSCALL. So when an application calls e.g. SYSCALL(mulu) then that call gets translated to a software interrupt. This means that while the system call is in progress, interrupts can not occur, and we avoid having a global interrupt disable bit (neither in the CPU nor in a hardware interrupt controller).

The question is, what happens when an INT instruction is encountered while processing an ISR ?

bernd-ulmann, 2020-09-17:
Why not have this in the CPU? 

* We only have one SR bit left and we will need this for user/kernel mode. So a EI/DI instruction would require a "hidden" bit within the CPU.

* EI/DI would require two additional control instructions.

* Later on we might want to have a way more complex interrupt controller that might take care of interrupt priorities, have a queue etc. At least then it will become a separate unit.

 Having an interrupt controller device solves all of these problems: We do not have to introduce yet another two instructions, we do not have to sacrifice the remaining control bit in the SR. We only need (at the moment) one address in IO space for the interrupt controller CSR (command and status register) which can be read and written to get all of the classic functionality with the option of introducing more complex things like interrupt priorities later etc. without the need to modify the CPU architecture. 

An INT instruction within an ISR is considered a rogue INT and will HALT the CPU.

MJoergen, 2020-09-17:
@bernd-ulmann . You have convinced me!

MJoergen, 2020-09-17:
I've implemented in hardware and tested it.

This still needs to be implemented in the emulator. I had a brief look, but became unsure. Perhaps @sy2002  or @bernd-ulmann  could make the necessary changes in the emulator ?

bernd-ulmann, 2020-09-17:
I'll take care of it. At which address is the CSR and which bit controls the interrupts. Does 1 enable or disable the interrupts? :-)

MJoergen, 2020-09-17:
Write 0x0001 to 0xFF50 to enable interrupts.

Write 0x0000 to 0xFF50 to disable interrupts.

bernd-ulmann, 2020-09-17:
Thank you for the extending sysdef.asm. I hope you are not mad at me as I renamed IC$ENABLE to IC$CSR for Control-Status-Register. This is just how I am used that such registers are called and it leaves us more freedom when we will find uses for the remaining 15 bits. :-) 

I have extended the emulator to support this new IO-device register. :-) I tested it with timer_test.asm. Loading and running this program does not cause any interrupts until one sets FF50 to 0001 with MEMORY/CHANGE. Setting this register to 0 stops all interrupts. :-)

By default interrupts are now disabled after start/reset.

closed 2020-09-17 by bernd-ulmann

MJoergen, 2020-09-18:
@bernd-ulmann : Great job! Thank you very much. The new name is much better :-)