sy2002/QNICE-FPGA #140: Proof-read the ISR best practices

sy2002, 2020-09-17:
I volunteered to collect all our recent best-practice findings around ISRs.

Here they are:

https://github.com/sy2002/QNICE-FPGA/blob/develop/doc/best-practices.md#interrupt-service-routines-isrs

@MJoergen and @bernd-ulmann please proof read and feel free to improve this file: `doc/best-practices.md`.

Bonus: I also volunteered and collected all our recent best-practice findings around register banks:

https://github.com/sy2002/QNICE-FPGA/blob/develop/doc/best-practices.md#register-banks

After that, please close this issue.

bernd-ulmann, 2020-09-17:
I really like the "be paranoid" part! :-) That is so true!

MJoergen, 2020-09-17:
I've made a few minor updates to the `best-practices.md` document. I'm satisfied now.

closed 2020-09-17 by sy2002