sy2002/QNICE-FPGA #145: World demo has a broken sprite when compiled with optimizations

MJoergen, 2020-09-23:
The demo program `world.c` does not work when compiled with -O3. One corner of the image is displayed at an incorrect position.

Compiling the program without optimizations gives the correct output.

This problem will hopefully be fixed when upgrading to a newer version of the compiler, see Issue #100

sy2002, 2020-09-23:
@MJoergen This is strange: I checked out the most recent branch and compiled without optimizations like this:

```
qvc world.c sprite.c -c99
```

And it looks like this:

[image]

What is really strange here is: Earlier today, when I wrote my celebration email to you and bernd and did the video, it worked fine. Might the reason be that I am still running the VHDL version from commit `f5886e6` ?

Off topic and maybe a small improvement/fix: Currently, you only check if a new UART key was pressed.

If you changed the line

```
      if (MMIO(IO_UART_SRA) & 1)
         break;
```

into something like this (warning, untested code):

```
      if (MMIO(IO_UART_SRA) & 1) || (MMIO(IO_KBD_STATE) & KBD_NEW_ANY)
         break;
```

then I could also use my USB keyboard to end the program. Since qtransfer works, I stopped using the UART ;-)

MJoergen, 2020-09-23:
> Might the reason be that I am still running the VHDL version from commit f5886e6 ?

Yes, this was a bug introduced because of some refactoring I did. It was fixed in commit 4165286.

> then I could also use my USB keyboard to end the program.

That should be possible now in the latest commit fc4b4ad.

sy2002, 2020-09-24:
Thank you for the hint! I re-synthesized today with the newest head in `dev-sprites` and now everything works like a charm - **including the mirroring of the globe**! 👍  Great work!

What I found remarkable though is: Even if I compile with optimizations (-O3): Everything *still* works like a charm, so is this issue still valid at all?

sy2002, 2020-09-25:
From Volker's email I learned, that if you use `-maxoptpasses=15`, the warning goes away that normally came when doing this:

```
 qvc world.c sprite.c -c99 -O3
```

So here are my test results:

Compiling with above-mentioned settings spits out a warning but works like a charm.

Compiling with this:
```
qvc world.c sprite.c -c99 -O3 -maxoptpasses=15
```

Works also like a charm and eliminates the warning. I will update the documentation about the option to increase the passes.

closed 2020-09-25 by sy2002