sy2002/QNICE-FPGA #148: Assembler lost ability to align within .lis file

sy2002, 2020-09-26:
In one of the latest updates (I cannot point to which one from the top of my head), the assembler lost its ability to align within the `.lis` file which makes reading the `.lis` file harder. The following screenshot shows the problem:

[image]

What seems to happen is: The code that comes behind a label is always printed with one space in the `.lis` file as opposed to the amount that was given in the source file.

bernd-ulmann, 2020-09-26:
...to be honest, I have seen this quite some time ago but always forgot to fix it... I will try to take care of it tomorrow or Monday at latest. :-)

bernd-ulmann, 2020-09-27:
Assembler listings now look qnice nice again. ;-)

closed 2020-09-27 by bernd-ulmann