sy2002/QNICE-FPGA #149: Comments are hard to read in .lis files

sy2002, 2020-09-28:
Hi Bernd, forgive me to add so many assembler issues these days. The reason is: By doing the best humanly thinkable textmode sprite, font-gfx and palette editor on the planet (dubbed "font_ed.asm") I re-started assembly programming on a quite intense level ;-)  This is why I am stumbling from time to time over small inconveniences. Here is one that I remember has been better in past - but maybe this is just how my brain works: "Früher war alles besser ;-)"  (and in reality: the past has been no way better like today ;-))

So here we go: This is how comments look in my source code:

[image]

And this is what the assembler does in the `.lis` file: The comments are harder to read to to the thrashing / wobbling:

[image]

No worries and ho hurries: This is just a minor inconvenience.

bernd-ulmann, 2020-09-28:
Here you go... ;-) I hope you like the new output format. :-)

sy2002, 2020-09-28:
Thanks for WARP-SPEED :-)

But this time, you "killed" my headline comments, as you are moving all of them to the end of the line. Look at this image: What you see here is a typical example of code, where I add a "headline" comment that is meant to describe, what the next section of code is doing. The headline comment is nicely aligned with the code itself. And then, behind the actual lines of assembly code, I am adding comments to describe what this line does:

[image]

Now: Your new version is throwing everything to the end and then the same code snippet looks like this:

[image]

I think a smart and easy to implement rule would be: 

**If a comment is sitting in a line alone with no other "thing" around it: Just leave it alone**

I am not insisting in this "fix" as this is really a "luxory problem" - so no problem with closing it and "WONTFIX".

bernd-ulmann, 2020-09-28:
The assembler leaves comments alone which start in column 1... *sigh* I will take care of it as soon as my phone conference will be over. :-)

bernd-ulmann, 2020-09-28:
Done... I hope you like the result. :-)

sy2002, 2020-09-28:
AWESOME 😄  Thank you!

closed 2020-09-28 by sy2002