sy2002/QNICE-FPGA #159: Emulator shows weird addresses in prompt

sy2002, 2020-10-11:
Here is an example: A breakpoint is set to $8A20. After the breakpoint is reached, the prompt shows $8BA0.

[image]

A quick check using `rdump` shows, that the PC is indeed at $8A20 (where it should be), so the prompt is obviously wrong.

bernd-ulmann, 2020-10-11:
Fixed (at least my simple breakpoint test worked). :-)

sy2002, 2020-10-11:
@bernd-ulmann Basic smoke test fails: It is not possible to single step using `step` any more with this fix: How to reproduce:  Set breakpoint and then try to `step`.

bernd-ulmann, 2020-10-11:
*sigh* I should not try to "fix" things so late at night... Please give it another try - it should work now... Sorry!

sy2002, 2020-10-11:
will gladly try, but you need to PUSH your local git first :-)

bernd-ulmann, 2020-10-11:
*ARGL* I did push but did not notice that it failed due to a merge required... I merged and pushed again... Time to go to bed, I think. :-)

sy2002, 2020-10-11:
Works :-) 👍

closed 2020-10-11 by sy2002