sy2002/QNICE-FPGA #161: Emulator kind of crashes after reset

sy2002, 2020-10-11:
This one is **not** urgent as the situation can only be provoked, when working with the `reset` command of the emulator.

Here it is:

[image]

sy2002, 2020-10-11:
sorry, wrong reproduction scheme. closing.

sy2002, 2020-10-11:
Well maybe it is the right reproduction scheme: @bernd-ulmann how should reset actually behave?

bernd-ulmann, 2020-10-11:
Reset does the following as of now:

- Clear all memory locations < IO_AREA_START.
- Clear all registers.
- Clear the shadow registers.
- Reset all statistics counters.
- Disable interrupts and clear a possibly pending interrupt.
- Set the switch register to 3.

sy2002, 2020-10-11:
OK, then I guess this is behaving as intended. Closing.

closed 2020-10-11 by sy2002