sy2002/QNICE-FPGA #163: Assembler listing output contains ghost instructions

bernd-ulmann, 2020-10-13:
The listing produced by the assembler contains "ghost" instructions. The cpu_test program shows this problem. This is the source:

                ABRA    E_MOVE_IMM_34, !1       ; Verify "absolute branch never" is not taken.
                ABRA    L_MOVE_IMM_34, 1        ; Verify "absolute branch always" is taken.
                HALT
E_MOVE_IMM_34   HALT
L_MOVE_IMM_34

...while this is the resulting listing file:

000001  0000  FF88  0005                              ABRA E_MOVE_IMM_34, !1           ; Verify "absolute branch never" is not taken.
000002  0002  FF80  0006                              ABRA L_MOVE_IMM_34, 1            ; Verify "absolute branch always" is taken.
000003  0004  E000                                    HALT
000004  0005  E000        E_MOVE_IMM_34               HALT
000005  0006              L_MOVE_IMM_34               HALT

EQU-list:
--------------------------------------------------------------------------------------------------------

Label-list:
--------------------------------------------------------------------------------------------------------
E_MOVE_IMM_34           : 0x0005    L_MOVE_IMM_34           : 0x0006    

 Not the spurious HALT instruction in line 000005...

bernd-ulmann, 2020-10-13:
Fixed it. 

A quick remark from my side: The assembler has somehow ;-} turned from a little tool into a rather ugly program. We should think about rewriting or at least fundamentally refactoring it in the future. :-)

sy2002, 2020-10-13:
@bernd-ulmann Thank you for finding and fixing this one. I assume this issue can be closed and closed it.