sy2002/QNICE-FPGA #166: Optimize random number generator and move to monitor

MJoergen, 2020-10-13:
The current implementation in `c/test_programs/rand.c` makes use of many large shift operations. This I believe can be refactored, simplified, and optimized. Possibly even rewrite in assembler.

And then move this function to the monitor, so it can be accessed via a SYSCALL().

sy2002, 2020-10-13:
Great idea! Please decide: V1.7 or V1.8 :-)

bernd-ulmann, 2020-10-13:
Which pseudorandom number generator would you suggest? A simple linear congruence generator of something more sophisticated? 

Just thinking "loud" - we could implement a (pseudo) random number generator as an IO-device where we could implement something quite sophisticated...? :-)

MJoergen, 2020-10-14:
@bernd-ulmann : I was planning (for now) on using the same Random Number Generator (RNG) as in the file `c/test_programs/rand.c`, i.e. something that generates the same sequence (given the same seed), and just optimize it and hand-code it in assembly. It is a simple linear congruence generator.

Yes, we could make something in hardware that is very close to a "true" RNG, but I think for the purposes of QNICE that is overkill.

sy2002, 2020-10-14:
> something that generates the same sequence (given the same seed), and just optimize it and hand-code it in assembly. It is a simple linear congruence generator.

A hand optimized assembly software solution and a `SYSCALL` that is part of the Monitor sound optimal for me.

MJoergen, 2020-10-16:
A hand-coded assembly version of the Random Number Generator is now implemented as part of the Monitor.

Currently, it is pushed to the branch `dev-rand`.

I've removed the C-implementation in `rand.c`/`rand.h`, and updated `maze2d` and `demo-sprite-ball` demo programs that both use random numbers.

I've done unit test on the generator itself, and a quick test of the `maze2d` program. Not yet tested on hardware.

@sy2002 and @bernd-ulmann : Could either of your please review, that I've correctly added to the monitor library ?

sy2002, 2020-10-16:
@MJoergen Sounds great! I will review that tomorrow.

MJoergen, 2020-10-17:
I've verified - using the test program `test_rand.c` - that the output is the same in the emulator as on the hardware.

sy2002, 2020-10-17:
I just did the code review I promised :-)

Everything good, including the way you added variables to the Monitor and the way you enhanced the C Monitor-Lib. A tiny thing was corrected:

I changed your
```
#define qmon_srand                  ((_qmon_fp)  QMON_EP_SRAND)
```
to
```
#define qmon_srand(x)               ((_qmon_fp)  QMON_EP_SRAND)(x)
```
and wonder, why your version ever worked :-)   (But it did, I know... ...probably VBCC is just a very forgiving compiler ;-))

Closing this issue as I consider it done.

closed 2020-10-17 by sy2002