sy2002/QNICE-FPGA #169: Test program int_test.asm fails in emulator

MJoergen, 2020-10-16:
Just using the latest on the `develop` branch.

I get the error in the emulator:

```
QMON> CONTROL/RUN ADDRESS=8000
Start
Rogue INT instruction with an ISR at address 8008. HALT!

391606 memory reads, 12322 memory writes and
213434 instructions have been executed so far:

INSTR ABSOLUTE         RELATIVE INSTR ABSOLUTE         RELATIVE
---------------------------------------------------------------
MOVE:            53373 (25.01%)	ADD :             1486 ( 0.70%)	
ADDC:                0 ( 0.00%)	SUB :             5939 ( 2.78%)	
SUBC:                0 ( 0.00%)	SHL :             1234 ( 0.58%)	
SHR :                0 ( 0.00%)	SWAP:                0 ( 0.00%)	
NOT :                0 ( 0.00%)	AND :            20056 ( 9.40%)	
OR  :                2 ( 0.00%)	XOR :             1978 ( 0.93%)	
CMP :            36342 (17.03%)	rsvd:                0 ( 0.00%)	
ctrl:            20864 ( 9.78%)	ABRA:                1 ( 0.00%)	
ASUB:                1 ( 0.00%)	RBRA:            61727 (28.92%)	
RSUB:            10431 ( 4.89%)	

         READ ACCESSES                       WRITE ACCESSES
MODE   ABSOLUTE         RELATIVE        MODE   ABSOLUTE         RELATIVE
------------------------------------------------------------------------
rx   :            81436 (23.70%) 	rx   :            82179 (23.91%)
@rx  :            27678 ( 8.05%) 	@rx  :             1886 ( 0.55%)
@rx++:           150494 (43.79%) 	@rx++:                4 ( 0.00%)
@--rx:                0 ( 0.00%) 	@--rx:                0 ( 0.00%)

Last addresses executed (last address first, then previous etc.):
8008	8006	8004	0BE0	0BDF	0BDE	0BD8	0BD6	
0BD5	0BDC	0C02	0C01	0F5D	0F5C	0F5B	0F59	

Register dump: BANK = 00, SR = ___N___1

	R00-R03:   807a 0000 0000 0000 
	R04-R07:   0000 0000 0000 0000 
	R08-R11:   801d 0c09 0000 0000 
	R12-R15:   0000 feeb 0011 8009 

Shadow registers (PC, SR, SP):

	SH00-SH03: 8006 0011 feeb 

[8008] Q> 
```

It seems that the PC does not get updated when executing an INT instruction.

bernd-ulmann, 2020-10-17:
Hmmm - it works for me:

➜  test_programs git:(develop) ✗ ../assembler/asm int_test.asm
qasm2rom: 123 ROM lines written.

➜  emulator git:(develop) ✗ qnice ../monitor/monitor.out

Simple QNICE-monitor - Version 1.7 (Bernd Ulmann, sy2002, September 2020)
-------------------------------------------------------------------------

QMON> MEMORY/LOAD - ENTER ADDRESS/VALUE PAIRS, TERMINATE WITH CTRL-E
0800000FA0080010803A080020FF900800300008080040E0BE08005080170800600FA0080070801D080080E0A00800900FA00800A080370800B0E0A10800C00FA00800D080390800E0E0A30800F0E0A2080100E0A10801100FA00801208075080130FF900801400008080150FF9008016000160801700FA00801808042080190FF900801A000080801B0E0400801C0E0000801D00FA00801E0804C0801F0FF900802000008080210E040080220E0000802300FA00802408056080250FF900802600008080270E040080280E00008029008370802A00FA00802B080600802C0FF900802D000080802E00DA00802F0E040080300E0000803100FA0080320806A080330FF900803400008080350E040080360E0000803708023080380802908039080310803A000530803B000740803C000610803D000720803E000740803F0000D080400000A0804100000080420004908043000530804400052080450005F080460004108047000420804800053080490000D0804A0000A0804B000000804C000490804D000530804E000520804F0005F080500005208051000450805200047080530000D080540000A0805500000080560004908057000530805800052080590005F0805A000490805B0004E0805C000440805D0000D0805E0000A0805F00000080600004908061000530806200052080630005F080640005008065000520806600045080670000D080680000A08069000000806A000490806B000530806C000520806D0005F0806E000500806F0004F08070000530807100054080720000D080730000A08074000000807500045080760006E0807700064080780000D080790000A0807A00000
QMON> CONTROL/RUN ADDRESS=8000
Start
End

QMON>

 Did I make some mistake here?

sy2002, 2020-10-17:
Hi Bernd, I just tested it on my machine: I can reproduce Michael's problem. Things to check:

1. Make sure that you are locally on the `develop` branch: `git checkout develop`
2. If in doubt: `git pull`
3. Try to recompile everything using

```
cd tools
./make-toolchain.sh 
```

or at least recompile the emulator using the local `make.bash` script in the `emulator` folder.

bernd-ulmann, 2020-10-17:
Strange... I was (and am) as you can see on my prompt in the develop branch. I tried to reproduce it again:
```
➜  emulator git:(develop) ✗ git pull
remote: Enumerating objects: 56, done.
remote: Counting objects: 100% (56/56), done.
remote: Compressing objects: 100% (23/23), done.
remote: Total 56 (delta 36), reused 53 (delta 33), pack-reused 0
Unpacking objects: 100% (56/56), done.
From https://github.com/sy2002/QNICE-FPGA
   5af31a2..67d95f5  develop    -> origin/develop
Updating 5af31a2..67d95f5
Fast-forward
 VERSIONS.txt                       |   11 +-
 demos/mandel_zoom.asm              |    2 +
 demos/q-tris.asm                   |    9 +
 demos/tile_ed.asm                  |   17 +-
 monitor/debug_library.asm          |    5 +
 monitor/fat32_library.asm          |    9 +-
 monitor/io_library.asm             |    3 +-
 monitor/qmon.asm                   |    1 +
 monitor/qtransfer.asm              |    1 +
 monitor/vga_library.asm            |    1 +
 qbin/fancy.out                     |   96 +-
 qbin/mandel_zoom.out               | 1156 ++++++-------
 qbin/q-tris.out                    | 2854 ++++++++++++++++----------------
 qbin/sdcard.out                    | 9556 ++++++++++++++++++++++++++++++++++++++++++++++++++++++-----------------------------------------------------
 qbin/tile_ed.out                   | 5122 ++++++++++++++++++++++++++++-----------------------------
 test_programs/fancy.asm            |    2 +
 test_programs/keyboard.asm         |    2 +
 test_programs/q-tris_perf_test.asm |    9 +
 test_programs/qtransfer.asm        |    1 +
 test_programs/sdcard.asm           |   15 +-
 test_programs/vga_clrscr.asm       |    1 +
 vhdl/alu.vhd                       |   15 +-
 vhdl/qnice_cpu.vhd                 |    3 +-
 23 files changed, 9500 insertions(+), 9391 deletions(-)
➜  emulator git:(develop) ✗ make.bash
```

Then I assembled the test program, too:
```
➜  test_programs git:(develop) ✗ ../assembler/asm int_test.asm
qasm2rom: 123 ROM lines written.
```

Loading and executing it yields the following:
```
➜  emulator git:(develop) ✗ qnice ../monitor/monitor.out

Simple QNICE-monitor - Version 1.7 (Bernd Ulmann, sy2002, September 2020)
-------------------------------------------------------------------------

QMON> MEMORY/LOAD - ENTER ADDRESS/VALUE PAIRS, TERMINATE WITH CTRL-E
0800000FA0080010803A080020FF900800300008080040E0BE08005080170800600FA0080070801D080080E0A00800900FA00800A080370800B0E0A10800C00FA00800D080390800E0E0A30800F0E0A2080100E0A10801100FA00801208075080130FF900801400008080150FF9008016000160801700FA00801808042080190FF900801A000080801B0E0400801C0E0000801D00FA00801E0804C0801F0FF900802000008080210E040080220E0000802300FA00802408056080250FF900802600008080270E040080280E00008029008370802A00FA00802B080600802C0FF900802D000080802E00DA00802F0E040080300E0000803100FA0080320806A080330FF900803400008080350E040080360E0000803708023080380802908039080310803A000530803B000740803C000610803D000720803E000740803F0000D080400000A0804100000080420004908043000530804400052080450005F080460004108047000420804800053080490000D0804A0000A0804B000000804C000490804D000530804E000520804F0005F080500005208051000450805200047080530000D080540000A0805500000080560004908057000530805800052080590005F0805A000490805B0004E0805C000440805D0000D0805E0000A0805F00000080600004908061000530806200052080630005F080640005008065000520806600045080670000D080680000A08069000000806A000490806B000530806C000520806D0005F0806E000500806F0004F08070000530807100054080720000D080730000A08074000000807500045080760006E0807700064080780000D080790000A0807A00000E
QMON> CONTROL/RUN ADDRESS=8000
Start
End

QMON>
```
? I see the problem that it does not print ISR_ABS etc. but I do not get the error with a rogue INT instruction... I will now first look why the output is not generated.

bernd-ulmann, 2020-10-17:
Interesting! NOW I can reproduce it but only after I recompiled the monitor?! Do you have any idea why recompiling the monitor changes the behaviour of this test program so fundamentally?

I will now investigate further why we get a rogue INT instruction. :-)

sy2002, 2020-10-17:
Can you try to rebuild everything:
```
cd tools
./make-toolchain.sh 
```

bernd-ulmann, 2020-10-17:
I can already reproduce the problem - recompiling the monitor did the trick although I do not understand why...

Give me a moment to investigate the rogue INT problem.

bernd-ulmann, 2020-10-17:
Phew - it should work now. Could you please test more intensively than I did (monitor runs, int_test runs as well...) as I am incredibly tired and on my way to bed for now. :-) Good night!

MJoergen, 2020-10-18:
SORRY for missing an important detail in my initial report: I had just recently updated the monitor so that after reset it enables the global interrupt register `FF50`.

Now the test program works (when interrupts are enabled) and it no longer crashes.

However, when global interrupts are disabled, the INT instructions appear disabled.

To reproduce:
```
mike@mike-desktop ~/git/sy2002/QNICE-FPGA/emulator (develop) $ ./qnice ../monitor/monitor.out 

Simple QNICE-monitor - Version 1.7 (Bernd Ulmann, sy2002, September 2020)
-------------------------------------------------------------------------

QMON> MEMORY/CHANGE ADDRESS=FF50 CURRENT VALUE=0001 NEW VALUE=0000
QMON> MEMORY/LOAD - ENTER ADDRESS/VALUE PAIRS, TERMINATE WITH CTRL-E
0800000FA0080010803A080020FF900800300008080040E0BE08005080170800600FA0080070801D080080E0A00800900FA00800A080370800B0E0A10800C00FA00800D080390800E0E0A30800F0E0A2080100E0A10801100FA00801208075080130FF900801400008080150FF9008016000160801700FA00801808042080190FF900801A000080801B0E0400801C0E0000801D00FA00801E0804C0801F0FF900802000008080210E040080220E0000802300FA00802408056080250FF900802600008080270E040080280E00008029008370802A00FA00802B080600802C0FF900802D000080802E00DA00802F0E040080300E0000803100FA0080320806A080330FF900803400008080350E040080360E0000803708023080380802908039080310803A000530803B000740803C000610803D000720803E000740803F0000D080400000A0804100000080420004908043000530804400052080450005F080460004108047000420804800053080490000D0804A0000A0804B000000804C000490804D000530804E000520804F0005F080500005208051000450805200047080530000D080540000A0805500000080560004908057000530805800052080590005F0805A000490805B0004E0805C000440805D0000D0805E0000A0805F00000080600004908061000530806200052080630005F080640005008065000520806600045080670000D080680000A08069000000806A000490806B000530806C000520806D0005F0806E000500806F0004F08070000530807100054080720000D080730000A08074000000807500045080760006E0807700064080780000D080790000A0807A00000
QMON> CONTROL/RUN ADDRESS=8000
Start
End
```

No messages printed between `Start` and `End`.

When running on hardware, the test program behaves the same regardless of whether the global interrupt register is enabled or disabled (i.e. regardless of the value written into `FF50`). So the question is what behaviour do we wish to have?

I suggest we change the emulator, so it works like the real hardware (rather than the other way around). It seems natural to me that the `INT` instruction works even when global interrupts are disabled.

bernd-ulmann, 2020-10-18:
Now that is a good question. Hmmm... I have no real opinion on that - somehow it seemed logical to disable any interrupts but you are right, we might still have INT instructions behaving as normal...

I will change the emulator accordingly. :-)

bernd-ulmann, 2020-10-18:
The INT instruction in the emulator works now regardless of the settings of $FF50. We now have to interrupt flags namely

gbl$hw_interrupt_request and
gbl$sw_interrupt_request.

Could you please retest if everything behaves as it should?

MJoergen, 2020-10-19:
Verified.

closed 2020-10-19 by MJoergen