sy2002/QNICE-FPGA #172: Hardware CPU: Save/Restore R8 to R12 in shadow registers during INT

sy2002, 2020-10-19:
Enhance CPU to save/restore R8 to R12 in shadow registers during INT

Context:
https://github.com/sy2002/QNICE-FPGA/issues/77#issuecomment-711412906

MJoergen, 2020-10-19:
I'm updating the `maze2d` game to include a timer interrupt based clock. It works in the emulator, and it would be great to have it working in hardware too. Just to let you know I hope you implement this soon :-D

sy2002, 2020-10-19:
I plan to do it today evening :-D

MJoergen, 2020-10-27:
I've implemented this now. I've verified no regression with the `cpu_test.asm` program, and I've extended the `int_test.asm` program to test the new feature.

The implementation is somewhat verbose, and can perhaps be refactored.

sy2002, 2020-10-28:
Thank you - due to my end of year crazyness it is unlikely that I contribute anything meaningful "before Christmas" :-)
And indeed: This does need refactoring - but this is not urgent, so I close this issue.

closed 2020-10-28 by sy2002