sy2002/QNICE-FPGA #174: emulator prints error and leaves terminal in "funny" state when using timer interrupt

MJoergen, 2020-10-19:


MJoergen, 2020-10-19:
The test program `test_isr.c` runs fine on the emulator, but when the user presses a key, the emulator ends with the following message:
```
[1, 0] timer could not be removed: Success
```
and the terminal no longer echoes key presses

bernd-ulmann, 2020-10-19:
I haven't compiled a C program for QNICE for such a long time that I really forgot how to do it. Could you write me a short how-to to reproduce the problem? :-)

sy2002, 2020-10-19:
Hi Bernd, how to compile in C: https://github.com/sy2002/QNICE-FPGA#programming-in-c

MJoergen, 2020-10-19:
These four commands will compile the C program

```
$ cd c
$ source setenv.source 
$ cd test_programs/
$ qvc test_isr.c 
```

bernd-ulmann, 2020-10-19:
That is really strange. The "Undefined error: 0" comes from the pthread_cancel()-call in timer.c... At the moment, I have no idea what causes this problem...?! Strange...

MJoergen, 2020-11-01:
I just ran the `strace` command with the emulator and got the following output:

```
read(0, "q", 1024)                      = 1
openat(AT_FDCWD, "/etc/ld.so.cache", O_RDONLY|O_CLOEXEC) = 3
fstat(3, {st_mode=S_IFREG|0644, st_size=134800, ...}) = 0
mmap(NULL, 134800, PROT_READ, MAP_PRIVATE, 3, 0) = 0x7f576277f000
close(3)                                = 0
openat(AT_FDCWD, "/lib/x86_64-linux-gnu/libgcc_s.so.1", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\3405\0\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0644, st_size=104984, ...}) = 0
mmap(NULL, 107592, PROT_READ, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7f5761d49000
mmap(0x7f5761d4c000, 73728, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x3000) = 0x7f5761d4c000
mmap(0x7f5761d5e000, 16384, PROT_READ, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x15000) = 0x7f5761d5e000
mmap(0x7f5761d62000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x18000) = 0x7f5761d62000
close(3)                                = 0
mprotect(0x7f5761d62000, 4096, PROT_READ) = 0
munmap(0x7f576277f000, 134800)          = 0
getpid()                                = 75924
tgkill(75924, 75935, SIGRTMIN)          = 0
futex(0x7f57625649d0, FUTEX_WAIT, 75935, NULL) = -1 EAGAIN (Resource temporarily unavailable)
dup(2)                                  = 3
fcntl(3, F_GETFL)                       = 0x2 (flags O_RDWR)
fstat(3, {st_mode=S_IFCHR|0620, st_rdev=makedev(0x88, 0), ...}) = 0
write(3, "[1, 0] timer could not be remove"..., 43[1, 0] timer could not be removed: Success
) = 43
close(3)                                = 0
exit_group(-1)                          = ?
+++ exited with 255 +++

```

MJoergen, 2020-11-01:
Once more, this time with `DEBUG` enabled:

```
read(0, "q", 1024)                      = 1
write(1, "timer: write access at address 0"..., 37timer: write access at address 0000.
) = 37
futex(0x7f1aa06a64c0, FUTEX_WAKE_PRIVATE, 1		Timer 0 triggered: INT = 809A.
) = 1
		Timer 0 instantiated. Interval = 0 s, 100000 ns
write(1, "\tTimer 0 was on, will now be dea"..., 42	Timer 0 was on, will now be deactivated.
) = 42
		Timer 0 triggered: INT = 809A.
		Timer 0 instantiated. Interval = 0 s, 100000 ns
openat(AT_FDCWD, "/etc/ld.so.cache", O_RDONLY|O_CLOEXEC) = 3
fstat(3, 		Timer 0 triggered: INT = 809A.
		Timer 0 instantiated. Interval = 0 s, 100000 ns
{st_mode=S_IFREG|0644, st_size=134800, ...}) = 0
mmap(NULL, 134800, PROT_READ, MAP_PRIVATE, 3, 0) = 0x7f1aa06cf000
		Timer 0 triggered: INT = 809A.
close(3		Timer 0 instantiated. Interval = 0 s, 100000 ns
)                                = 0
openat(AT_FDCWD, "/lib/x86_64-linux-gnu/libgcc_s.so.1", O_RDONLY|O_CLOEXEC		Timer 0 triggered: INT = 809A.
) = 3
		Timer 0 instantiated. Interval = 0 s, 100000 ns
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\3405\0\0\0\0\0\0"..., 832) = 832
fstat(3, 		Timer 0 triggered: INT = 809A.
		Timer 0 instantiated. Interval = 0 s, 100000 ns
{st_mode=S_IFREG|0644, st_size=104984, ...}) = 0
mmap(NULL, 107592, PROT_READ, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7f1a9fc99000
		Timer 0 triggered: INT = 809A.
mmap(0x7f1a9fc9c000, 73728, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x3000		Timer 0 instantiated. Interval = 0 s, 100000 ns
) = 0x7f1a9fc9c000
mmap(0x7f1a9fcae000, 16384, PROT_READ, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x15000		Timer 0 triggered: INT = 809A.
) = 0x7f1a9fcae000
		Timer 0 instantiated. Interval = 0 s, 100000 ns
mmap(0x7f1a9fcb2000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x18000) = 0x7f1a9fcb2000
		Timer 0 triggered: INT = 809A.
		Timer 0 instantiated. Interval = 0 s, 100000 ns
close(3)                                = 0
		Timer 0 triggered: INT = 809A.
		Timer 0 instantiated. Interval = 0 s, 100000 ns
mprotect(0x7f1a9fcb2000, 4096, PROT_READ) = 0
munmap(0x7f1aa06cf000, 134800		Timer 0 triggered: INT = 809A.
		Timer 0 instantiated. Interval = 0 s, 100000 ns
)          = 0
getpid()                                = 76184
tgkill(76184, 76200, SIGRTMIN)          = 0
write(1, "timer: write access at address 0"..., 37timer: write access at address 0001.
) = 37
write(1, "\tTimer 0 was on, will now be dea"..., 42	Timer 0 was on, will now be deactivated.
) = 42
dup(2)                                  = 3
fcntl(3, F_GETFL)                       = 0x2 (flags O_RDWR)
fstat(3, {st_mode=S_IFCHR|0620, st_rdev=makedev(0x88, 0), ...}) = 0
write(3, "[1, 0] timer could not be remove"..., 43[1, 0] timer could not be removed: Success
) = 43
close(3)                                = 0
exit_group(-1)                          = ?
+++ exited with 255 +++

```