sy2002/QNICE-FPGA #192: Assembler does not allow semicolon inside a string

sy2002, 2020-11-28:
The assembler does not allow semicolon inside a string.

If you copy the following snippet into an `.asm` file and assemble it, then you are getting five warnings and the strings that contain a semicolon `;` are not assembled.

```
STR_LS_START0   .ASCII_W "; ======================================================================="
STR_LS_START1   .ASCII_W ";"
STR_LS_START2   .ASCII_W "; -----------------------------------------------------------------------"
STR_LS_START3   .ASCII_W "; DO NOT MODIFY! Autogenerated with TileEd V2.0"
STR_LS_CMT      .ASCII_W ";"
STR_LS_SPACE    .ASCII_W " "
STR_LS_PAL      .ASCII_W "PALETTE"
STR_LS_FONT     .ASCII_W "FONT"
STR_LS_TILE     .ASCII_W "TILE"
STR_LS_EQU      .ASCII_W ".EQU"
STR_LS_DW       .ASCII_W ".DW"
```

The output is:

```
assemble: Line 1: WARNING - Did not find closing double quote!
assemble: Line 2: WARNING - Did not find closing double quote!
assemble: Line 3: WARNING - Did not find closing double quote!
assemble: Line 4: WARNING - Did not find closing double quote!
assemble: Line 5: WARNING - Did not find closing double quote!
```

And the first few lines of the `.lis` file are:

```

*** Line 1: WARNING - Did not find closing double quote! ***
000001  0000  0000        STR_LS_START0               .ASCII_W                         ; ======================================================================="

*** Line 2: WARNING - Did not find closing double quote! ***
000002  0001  0000        STR_LS_START1               .ASCII_W                         ;"

*** Line 3: WARNING - Did not find closing double quote! ***
000003  0002  0000        STR_LS_START2               .ASCII_W                         ; -----------------------------------------------------------------------"

*** Line 4: WARNING - Did not find closing double quote! ***
000004  0003  0000        STR_LS_START3               .ASCII_W                         ; DO NOT MODIFY! Autogenerated with TileEd V2.0"

*** Line 5: WARNING - Did not find closing double quote! ***
000005  0004  0000        STR_LS_CMT                  .ASCII_W                         ;"
```

bernd-ulmann, 2020-11-28:
We definitely should think about a complete rewrite of the assembler in the not too far future, or at least about a major refactoring. Things like this problem have really gotten nasty to fix.

I will try to think of an (easy) fix...

bernd-ulmann, 2020-11-28:
OK, the assembler can now take care of strings containins a ';'. 

The solution has a tiny problem: Lines are typically processed by a pretty print routine (which was once called expand_tabs() as this was initially all it did - expand tabs... ;-) Since this routine has turned into a rather ugly monster, it has been renamed to transmogrify() and we should definitely think about a refactoring of the assembler, the code quality has not increased in the last 13 years...). This is now disabled for lines containing strings and comments. The effect on the .lis-files is small but should, nevertheless, be noted.

closed 2020-11-28 by bernd-ulmann