sy2002/QNICE-FPGA #193: Emulator's dump is not always showing the ASCII decoding

sy2002, 2020-11-29:
Look at this `dump` command in the emulator:

```
[8F12] Q> dump $9107, $9110
	
9107: 0000 0030 0030 0030 0031 0000 0000 0000 	   0 0 0 1      
910f: 0000 0000 
```

Everything is correct. `0x0030` is the hex representation of the ASCII value of the character `0` and `0x0031` of the character `1`, so the `0 0 0 1` shown on the right side is correct.

Now look at this `dump` comand with a slightly modified start address compared to above:

```
[8F12] Q> dump $9108, $910f
	
9108: 0030 0030 0030 0031 0000 0000 0000 0000 
```

The only difference to above is, that the first memory location has a value != 0 (i.e. `0x0030`). 

This (or some other obscure reason) leads to the emulator not printing the decoded `0 0 0 1`  any more.

bernd-ulmann, 2020-11-29:
I have problems reproducing this strange effect. I wrote a simple test program and loaded it into the emulator:

```
        .ORG    $8000
        HALT
        MOVE    R0, R0
        .ASCII_W    "0001"
```

Disassembling this, starting from various start addresses looks like this:

```
[0000] Q> load test.out
[0000] Q> dump 8000, 8020

1f40: 0000 0000 0000 0000 0000 0000 0000 0000
1f48: 0000 0000 0000 0000 0000 0000 0000 0000
1f50: 0000 0000 0000 0000 0000
[0000] Q> dump $8000,$8020

8000: e000 0000 0030 0030 0030 0031 0000 0000 	     0 0 0 1
8008: 0000 0000 0000 0000 0000 0000 0000 0000
8010: 0000 0000 0000 0000 0000 0000 0000 0000
8018: 0000 0000 0000 0000 0000 0000 0000 0000
8020: 0000
[0000] Q> dump $8001, $8020

8001: 0000 0030 0030 0030 0031 0000 0000 0000 	   0 0 0 1
8009: 0000 0000 0000 0000 0000 0000 0000 0000
8011: 0000 0000 0000 0000 0000 0000 0000 0000
8019: 0000 0000 0000 0000 0000 0000 0000 0000
[0000] Q> dump $8002, $8020

8002: 0030 0030 0030 0031 0000 0000 0000 0000 	 0 0 0 1
800a: 0000 0000 0000 0000 0000 0000 0000 0000
8012: 0000 0000 0000 0000 0000 0000 0000 0000
801a: 0000 0000 0000 0000 0000 0000 0000
[0000] Q> dump $8003, $8020

8003: 0030 0030 0031 0000 0000 0000 0000 0000 	 0 0 1
800b: 0000 0000 0000 0000 0000 0000 0000 0000
8013: 0000 0000 0000 0000 0000 0000 0000 0000
801b: 0000 0000 0000 0000 0000 0000
```

Which program did you load to trigger this behaviour? Maybe mine is too short?

bernd-ulmann, 2020-11-29:
Wow! That is strange. I could now reproduce it by changing the .ORG in the litte test program I posted to 0x9114. Issuing your exact dump commands shows the problem... I will take care of this now. :-)

bernd-ulmann, 2020-11-29:
Solved... The easiest solution to fix this problem was to change the dump command so that it always dumps an integer multiple of eight words.

closed 2020-11-29 by bernd-ulmann