sy2002/QNICE-FPGA #200: 2 bugs in qnice-intro.pdf

rofl0r, 2024-03-26:
- there's a typo "rouine" instead of routine
- page 26 talks about `INT <dst, op>`, then refers to "source operand"

sy2002, 2024-03-26:
Thank you @rofl0r ! @bernd-ulmann can you please fix this (in the master branch).

bernd-ulmann, 2024-03-28:
Done. :-) Thank you very much for letting me know. :-)

closed 2024-03-28 by bernd-ulmann

rofl0r, 2024-03-28:
actually, sorry for bothering you about this - i only noticed that you actually care about typos (last commit 3 years ago)

rofl0r, 2024-03-28:
hmm, actually the last commit from 3 years ago is still the last commit... /me scratches beard... then shrugs `*whatever*`

sy2002, 2024-03-28:
@bernd-ulmann Thank you Bernd for your fast reaction. Did you push your local commit to GitHub already?

bernd-ulmann, 2024-03-28:
Dear Mirko - 
I fixed it and immediately committed it to the master.
 Have a good night! :-) Yours, Bernd. :-)

rofl0r, 2024-03-29:
committed != pushed

sy2002, 2024-03-31:
Yes, I can see the fix in the master branch now. Thank you both.

rofl0r, 2024-03-31:
thanks! very nice cpu design btw, reminds me of knuth's mmix. only thing i miss are mul and div instructions. implementing those in terms of bit ops is pretty costly, otoh implementing bswap using shifts isn't. i hope if there is every a V2 of this design, it comes with div and mul, and maybe even mod. (i suspect you left these out for TTL impls).

sy2002, 2024-03-31:
@rofl0r Thank you for your feedback :-) @bernd-ulmann the inventor of the ISA will for sure be glad to here that :-) The "Monitor Operating System" is offering mul and div operations that are implemented in software and that can be invoked using SYSCALL(mulu, 1) (for unsigned for example) and even SYSCALL(mulu32, 1) for 32-bit multiplications. See:

https://github.com/sy2002/QNICE-FPGA/blob/master/monitor/math_library.asm#L30

and

https://github.com/sy2002/QNICE-FPGA/blob/master/doc/monitor/doc.pdf

rofl0r, 2024-03-31:
nice, reading up in a moment. if i understood correctly, the opcode space providing the interrupt routines should still have a few opcodes left that could be used to implement z80-style "extended" opcodes that all use a common prefix. like in z80, it would appear as an afterthought, but it's still better than nothing. that would be a QNICE MMX, i suppose.

bernd-ulmann, 2024-03-31:
Hi - first of all, thank you very much for your mails and for spotting the typos/errors in the introduction. I am really happy that you like the architecture - it is a descendant of a 32 bit architecture which I developed as a student in the early 1990s (with a TTL implementation), called NICE ("NICE Is Charmingly Elegant"). The plan for QNICE was to be "quite NICE" :-) and simpler to implement in TTL, which, unfortunately, I never found time to do so, but then Mirko did this great FPGA-implementation.

Multiplication and division (signed as well as unsigned) are implemented as an "IO device", so to speak, the EAE ("Extended Arithmetic Element") - this can be seen quite easily in the emulator (qnice.c). This idea was stolen from early PDP11 systems like the 11/40. This can be easily modified to include modulus etc. without having to sacrifice opcodes.

Regarding the opcodes, you are correct. :-)